ZipsRacingElectric/VCU-ZR25-Senior-Design #15: Dashboard Interface Task

purifiedbagels, 2025-02-21:


bglen, 2025-02-21:
For basic functionality:
- the VCU will not communicate with the MoteC LCD display. Instead, any data this display needs, it will grab it from the data-logging messages on the CAN bus through its own programming.

**Start button input (3.3V_START_BUTTON_INPUT):**
- used for entering Ready to Drive mode

**Dashboard VCU Fault LED (3.3V_DASH_INPUT_1):**
- There is an LED on the dash that can light up whenever there is a VCU fault we want to display. 3.3V_DASH_INPUT_2 technically should be configured as a digital output for this. Whenever there is a fault that triggers the ~3.3V_VCU_FAULT opening the shutdown loop relay, we should turn this LED on also so we can debug what happened.
- The reason we don't send the ~3.3V_VCU_FAULT signal directly to this LED is because there might be a minor fault (APPS for example), that we want to trigger the LED but don't want the shutdown loop relay opening and causing the car to fall out of high voltage.

**DRS enable / disable button (3.3V_DASH_INPUT_2):**
- This allows the driver to toggle the drag reduction system. The VCU will have to keep track of the DRS enable state, and change it when this button is pressed. 
- The DRS system is on a separate board connected to the CAN bus. The VCU will have to send a CAN message over the bus when the DRS enable state changes. (TBD) I don't know if this message is supposed to get sent at a regular interval, or only when the state changes (and some kind of handshake is done).
- The DRS state can be toggled regardless of the VCU state.

**Torque limit Increase button (3.3V_DASH_INPUT_3):**
- When pressed the torque limit variable increases by some percentage (lets say 10% for now)
- The torque limit variable will be part of the torque control handler, is a percentage 0 to 100% that gets passed to whatever algorithm is calculating the motor torques.
- torque limit variable should be limited between 10 to 100%. It might just be an integer that is interpreted by the algorithm as a percentage.

**Torque limit decrease button (3.3V_DASH_INPUT_4):**
- When pressed the torque limit variable decreases by some percentage (lets say 10% for now)

purifiedbagels, 2025-02-22:
@bglen I'd assume that if the shutdown loop opens externally, we should also turn on DASH_INPUT_2?
Actually thinking about this more, how does this work with configuring input 2 as an output for the VCU fault but as an input otherwise. It can only be one at a time, so do we assume that the dashboard will just keep the LED on even when we have to switch it back to an input, where the signal would go low?

purifiedbagels, 2025-02-22:
Yeah I'm a bit confused. The schematics show that the dash inputs are actual inputs, but 1 is an output and 2 sometimes is?

bglen, 2025-02-24:
That was a mistake, under **Dashboard VCU Fault LED (3.3V_DASH_INPUT_1):** I meant to write 3.3V_DASH_INPUT_1

The schematic calls these signals inputs, but in reality I should update it and call them GPIO as they can serve a more general purpose other than button inputs

purifiedbagels, 2025-02-24:
@tetrahedron341 Issue closed besides CAN

closed 2025-04-02 by purifiedbagels