ZipsRacingElectric/VCU-ZR25-Senior-Design #17: Yaw Rate Controller/Torque Vectoring Algorithm

purifiedbagels, 2025-02-21:
@bglen Not really sure what this entails and if this is on your end or Tetra and I's

bglen, 2025-02-21:
So for a first iteration:

This can be a placeholder for the future TV algorithm by simply scaling the APPS pedal percentage by a max torque variable to get the torque request. This can then be applied to each motor equally.

Next iteration:
- We will want to have multiple algorithms that we can select. One would essentially be the simple algorithm mentioned above, which we can call a limp mode
- Next we would have the torque vectoring algorithm. This probably will be some function or class with a PID implementation. 
- For the handler that selects the algorithm, it would be a good idea to pass each algorithm a struct containing all the vehicle data the algorithm needs to compute. We could make this struct the same for all algorithms, and simpler controllers just wouldn't use all the data.

bglen, 2025-02-21:
From engineering requirements:

1.1.9 The system shall provide a method of switching between different “control modes” for the powertrain, which can be selected by the driver (See Table 27 in Appendices).

bglen, 2025-02-22:
[image]

First attempt at the kind of data we would want to pass to the control algorithm in an input struct (or with multiple structs). I've put this spreadsheet Torque Control Struct.xlsx on the DT10 drive and will update it as we figure it out.

I can also see an initialization struct being used to send constant data needed by the control algorithms (things like constant vehicle parameters, like mass). Could also just be hard coded.

purifiedbagels, 2025-02-24:
instead of creating a new struct, it's likely that this will all be in the vehicledata struct that the algorithm can get

purifiedbagels, 2025-04-02:
closing because this is no longer specifically different from the #19

closed 2025-04-02 by purifiedbagels