ZipsRacingElectric/VCU-ZR25-Senior-Design #21: Vehicle Data Task

bglen, 2025-02-27:
Not entire sure what the plan to implement this is, so edit as necessary:

1. Data collection via CAN bus and input drivers
2. Data filtering / saturation

bglen, 2025-03-16:
**GPS**
Need to capture and record data from the GPS to be used by the control system

[image]

Data that is relevant to us and should be grabbed from the messages:
- Speed (needs to be converted to m/s)
- GPS Status
- Heading motion (convert to rad)
- Heading vehicle (convert to rad)
- Z angle rate (this is yaw rate, convert to rad/s)
- X acceleration (convert to m/s^2)
- Y acceleration (convert to m/s^2, and invert to put it in the control system coordinate system)

Notes:
- some of the data will be signed
- these messages will be sent at 200 Hz from GPS
- It is very important that we keep track of the GPS status in the 2nd message. It may take like 20 seconds after the car is turned on for the GPS to get a good lock, and then we can reliably use the data to drive. The VCU should trigger a fault that turn the fault LED on the dash on until the GPS lock is found (fault LED is triggered by 3.3V_DASH_INPUT_1)

[ECUMaster GPStoCAN V2 Manual (1).pdf](https://github.com/user-attachments/files/19274239/ECUMaster.GPStoCAN.V2.Manual.1.pdf)

bglen, 2025-03-16:
**MoTeC VIM**

[todo]

bglen, 2025-03-16:
**Battery Management System**

bglen, 2025-03-16:
**Initialization of vehicle data values**

Before data is populated from sensors, we could:
1. initialize everything to 0. We would have to avoid dividing by zero from calling functions before real data gets populated (ex: enabling control algorithm and passing 0 data to it)
2. initialize everything to NaN (from math.h). Then our math functions, specifically torque handling task would check if passed data is NaN via isnan() and handle the exception. We could also pass NaN when a sensor faults (ex: out of bounds)

purifiedbagels, 2025-03-17:
A vehicledata task has been started that all tasks call to to update the data, will need to be updated based on this issue and I think a lot of what is left to store is from CAN. Fault mgmt task can check this data for faults